Cosys-Lab/Cosys-AirSim #1: Unable to Compile Blocks Project on Ubuntu 20.04

Rong-Tao, 2023-10-26:
## Environment:
- Ubuntu 20.04
- Unreal 4.24.4
- 10.0.0-4ubuntu1 
## Steps to reproduce
### 1. Airsim Compilation
```bash
#!/bin/bash
unzip Cosys-AirSim-main.zip
cd /home/rong/cosys_airism/Cosys-AirSim-main
find . -type f -name "*.sh" -exec chmod +x {} \;
./setup.sh
./build.sh
```
I do get a lot of warnings but the code compiles successfully. UE4.24.4 compiled successfully as well beforehand.

## 2. Compile Blocks project
Then I run, 
```bash
#!/bin/bash
./UnrealEngine/Engine/Binaries/Linux/UE4Editor /home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Blocks.uproject
```
The log is the following
```
Increasing per-process limit of core file size to infinity.
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogInit: Display: Running engine for game: Blocks
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 26 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.031481
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +8:00, Platform Override: ''
LogPluginManager: Mounting plugin [lots of plugins]
LogPluginManager: Mounting plugin AirSim
Launching UnrealBuildTool... [/home/rong/cosys_airism/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe -Mode=QueryTargets -Project="/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Blocks.uproject" -Output="/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Intermediate/TargetInfo.json" -progress]
LogInit: Warning: Incompatible or missing module: Blocks
LogInit: Warning: Incompatible or missing module: AirSim
LogInit: Initializing SDL.
LogInit: Initialized SDL 2.0.10 revision: 12952 (hg-12952:bc90ce38f1e2) (compiled against 2.0.10)
LogInit: Using SDL video driver 'x11'
```
Then I get a prompt from UE4:
```
The following modules are missing or built with a different engine version: Blocks, Airsim, rebuild?
```
Click yes, and I get the following
```
STUBBED: FDesktopPlatformLinux::GetNativeFeedbackContext at /home/rong/cosys_airism/UnrealEngine/Engine/Source/Developer/DesktopPlatform/Private/Linux/DesktopPlatformLinux.cpp:449 (GetNativeFeedbackContext)
Running /home/rong/cosys_airism/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Blocks.uproject" -TargetType=Editor -Progress -NoHotReloadFromIDE

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/cosys_airism/UnrealEngine/Engine ~/cosys_airism/UnrealEngine/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/home/rong/cosys_airism/UnrealEngine).
@progress push 5%
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Blocks.uproject" "/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
Reflection code generated for BlocksEditor in 2.901007 seconds
@progress pop
Building BlocksEditor...
[Upgrade]
[Upgrade] Using backward-compatible build settings. The latest version of UE4 sets the following values by default, which may require code changes:
[Upgrade]     bLegacyPublicIncludePaths = false                 => Omits subfolders from public include paths to reduce compiler command line length. (Previously: true).
[Upgrade]     ShadowVariableWarningLevel = WarningLevel.Error   => Treats shadowed variable warnings as errors. (Previously: WarningLevel.Warning).
[Upgrade]     PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs   => Set in build.cs files to enables IWYU-style PCH model. See https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/IWYU/index.html. (Previously: PCHUsageMode.UseSharedPCHs).
[Upgrade] Suppress this message by setting 'DefaultBuildSettings = BuildSettingsVersion.V2;' in BlocksEditor.Target.cs, and explicitly overriding settings that differ from the new defaults.
[Upgrade]
Performing 7 actions (12 in parallel)
[4/7] Compile Module.AirSim.1_of_3.cpp
[3/7] Compile Module.AirSim.gen.cpp
[2/7] Compile Module.AirSim.3_of_3.cpp
[1/7] Compile Module.AirSim.2_of_3.cpp
clang++: error: no such file or directory: '@/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.gen.cpp.o.rsp'
clang++: error: no such file or directory: '@/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.1_of_3.cpp.o.rsp'
clang++: error: no input files
clang++: error: no such file or directory: '@/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.2_of_3.cpp.o.rsp'
clang++: error: no input files
clang++: error: no input files
clang++: error: no such file or directory: '@/home/rong/cosys_airism/Cosys-AirSim-main/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.3_of_3.cpp.o.rsp'
clang++: error: no input files
LogInit: Warning: Still incompatible or missing module: Blocks
LogInit: Warning: Still incompatible or missing module: AirSim
LogCore: Engine exit requested (reason: EngineExit() was called)
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module DesktopPlatform (44)
LogModuleManager: Shutting down and abandoning module PlatformCryptoOpenSSL (42)
LogModuleManager: Shutting down and abandoning module PlatformCryptoTypes (40)
LogModuleManager: Shutting down and abandoning module PlatformCrypto (38)
LogModuleManager: Shutting down and abandoning module AnimationModifiers (36)
LogModuleManager: Shutting down and abandoning module PropertyEditor (35)
LogModuleManager: Shutting down and abandoning module AudioEditor (32)
LogModuleManager: Shutting down and abandoning module TextureCompressor (30)
LogModuleManager: Shutting down and abandoning module RenderCore (28)
LogModuleManager: Shutting down and abandoning module Landscape (26)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (24)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (22)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (20)
LogModuleManager: Shutting down and abandoning module Renderer (18)
LogModuleManager: Shutting down and abandoning module Engine (16)
LogModuleManager: Shutting down and abandoning module CoreUObject (14)
LogModuleManager: Shutting down and abandoning module NetworkFile (12)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
LogModuleManager: Shutting down and abandoning module StreamingFile (8)
LogModuleManager: Shutting down and abandoning module SandboxFile (6)
LogModuleManager: Shutting down and abandoning module PakFile (4)
LogModuleManager: Shutting down and abandoning module RSA (3)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
```
And UE4 showed an error prompt telling that "Could not be compiled, try rebuilding manually from source"

- I didn't copy the Plugin folder into the Blocks as the Plugin is already in that folder.
- I managed to compile UE4.27+Airsim and UE5.2.0+ Coloseum and get data from there.

WouterJansen, 2023-10-27:
have you tried running the [update_from_git.sh](https://github.com/Cosys-Lab/Cosys-AirSim/blob/main/Unreal/Environments/Blocks/update_from_git.sh) script before starting the build? 
Normally the build.sh script indeed does this automatically for the Blocks environment at the end but could try doing it again to make sure. 

I will try to replicate your build but currently on holiday so won't be able to for at least 3 weeks.

WouterJansen, 2023-11-17:
So I did a fresh install of the 4.24.4 UnrealEngine fork and Cosys-AirSim with the same commands as you on a Ubuntu 20.04.4. Unfortunately I am not able to replicate the issue as my editor loads up fine and I can run the simulation. The only thing different that I did was run ./update_from_git.sh in the Blocks folder after AirSim had completed its initial build script.

Rong-Tao, 2023-11-29:
Yes, I manage to compile the project with the process you described! thanks a lot for your help. By the way, just asking if you intend to migrate your work to be compatible with UE5 (city samples)?

WouterJansen, 2023-11-29:
If we find time and budget to migrate to UE5 we surely will! We would love to use the procedural generation tooling it offers and all the other benefits of a more modern engine.

closed 2023-11-29 by WouterJansen

Rong-Tao, 2023-12-08:
>
>
> Dear Wouter,
>
> Hi, it's me again. I have a question regarding the compilation of
> StaticMeshActors in Unreal Engine (UE). I am currently using UE5 along with
> Airsim (Colosseum) for my project. During my work on instance segmentation,
> I realized that many objects are rendered using Foliage, leading to them
> being considered as the same object.
>
> To address this, I created a UE blueprint and looped through all the
> instances, converting them back to mesh. However, I encountered a problem
> in Editor mode. When I ran the following function:
>
def seg_reid(client):
object_names = client.simListSceneObjects(name_regex = "StaticMeshActor_.*")
id = 0
for ob in object_names:
print(id, ob)
client.simSetSegmentationObjectID(ob, id % 256)
id += 1

 I was able to capture around 300 meshes and change their segmentation IDs
(segid), ranging from StaticMeshActor_0 to StaticMeshActor_351.

However, issues arose when I packaged the project and ran it. I discovered
that I only got about 4 names from object_names. Does UE automatically
optimize these aspects? It seems that even if I perform certain operations
on the Airsim side, the Unreal Engine side might still make it impossible
to segment all these meshes.

> Message ID: ***@***.***
> .com>
>

WouterJansen, 2023-12-08:
Not sure what would be wrong with your code. But I do not know how Colloseum does some of their functionality so I think you best off asking them directly.