Cosys-Lab/Cosys-AirSim #22: raise error C2220 when I run build.cmd from the command line.

hollywuzh, 2024-08-13:
I followed the guide (https://cosys-lab.github.io/install_windows/) and tried to build Cosys-AirSim. 
In the step "Run build.cmd from the command line". The command prompt raise error as blow. 
[image]
The similar sceniro can be found in (https://github.com/Microsoft/AirSim/issues/76). However, I check the encoding of "Half.h" and it has no problem. So how can I solve this problem? Thanks.

WouterJansen, 2024-08-14:
Can you try to re-save _Half.h_ as UTF-8 with BOM and try to compile again? I know Notepad++ can easily do this: 
[image]

hollywuzh, 2024-08-15:
The problem has been solved. I re-save Half.h\BFloat16.h\GenericPacketMathFunctions.h as UTF-8 with BOM in VSCode. Thanks!

closed 2024-08-16 by WouterJansen