Cosys-Lab/Cosys-AirSim #24: cant Run Ros

reza-shahriari, 2024-08-21:
hi,
i did every thing like [this link](https://cosys-lab.github.io/ros_python/) i mean just catkin_make in the python_ws and then source the path.
now i want to run my ros and get images or whatever i want. the problem is when i run rosrun airsimros airsim_publish.py i see this error:
```
Traceback (most recent call last):
  File "/home/iust/Documents/Unreal_Projects/Cosys-AirSim-5.4/ros/python_ws/src/airsimros/scripts/airsim_publish.py", line 1575, in <module>
    airsim_publish(client, use_route, route_rosbag, merged_rosbag, generate_gt_map, tf_static,
  File "/home/iust/Documents/Unreal_Projects/Cosys-AirSim-5.4/ros/python_ws/src/airsimros/scripts/airsim_publish.py", line 663, in airsim_publish
    pose_publisher = rospy.Publisher(pose_topic, PoseStamped, queue_size=1)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 842, in __init__
    super(Publisher, self).__init__(name, data_class, Registration.PUB)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 131, in __init__
    raise ValueError("topic name is not a non-empty string")
ValueError: topic name is not a non-empty string
```
I just comment everywhere pose_publisher was and then saw this error:
`[ERROR] [1724240139.730937]: vehicle 'airsimvehicle' could not be found.`
would you help me please

WouterJansen, 2024-08-22:
Can you show me your settings.json configuration and ROS launch file?

reza-shahriari, 2024-08-22:
the setting file: http://jsonbrulob.com/1276180054588383232
i have used PublishToRos for one of my cameras. 
i am unfamiliar with ROS, I just ran a roscore and used `rosrun python_ws airsim_publish.py`

WouterJansen, 2024-08-23:
Your link does not work for me. The AirSim Publish ROS Node that we made (airsim_publish.py) uses a lot of settings and arguments to configure. You need to match the configuration in ROS with that of AirSim in your settings file. For example to define the name of the vehicle, sensors, ROS-topics to publish, ... Hence a [ROS Launch](http://wiki.ros.org/roslaunch) file is easier to use. 

Once you have configured the launch file (we provide one [here](https://github.com/Cosys-Lab/Cosys-AirSim/blob/main/ros/python_ws/src/airsimros/launch/airsim_publish.launch) you can start the node with `roslaunch airsim airsim_publish.launch`

reza-shahriari, 2024-08-23:
Thanks a lot, i'll try it. it could be nice if you update the documentation a bit ;)

closed 2024-08-23 by reza-shahriari