Cosys-Lab/Cosys-AirSim #25: slow image capturing

reza-shahriari, 2024-08-21:
Is there a way that we can make image capture process run faster? it seems to be sequential instead of parallel specially when you capture tow images instead of one the image, fps get lower and lower!
the environment with many object and features running in real time but when u start get image its slow!
i have a real strong computer, the airsim is not using all of my resources. i mean to many cpu cores and a lot of gpu are not in use but the process is slow!
it will be nice if you help me change it to a faster way, if you can't impelement your self just tell me the roadmap and ill try to do it myself.
thanks for your nice work :)
 
 i know i can use noDisplay or turn of windows or even smaller images and then it will be faster but i want to solve the problem not just working with it

WouterJansen, 2024-08-22:
https://github.com/microsoft/AirSim/pull/3018
https://github.com/microsoft/AirSim/issues/4101

It is a known issue that GPU/CPU memory transfer and sequential image rendering are slow in (Cosys-)AirSim/Colloseum.

There are a few things that have been tried without success but there does not seem to be a magic thing that can be done to make it faster, unfortunately.

reza-shahriari, 2024-08-22:
Thank you for your answer. after asking the questions I have searched a bit and i have found that someone used `FRHIGPUTextureReadback` and said it's faster for this task... i will try it and share the results.

reza-shahriari, 2024-09-28:
I have tested some other capturing methods such as Streaming cameras but nothing worked as I wanted!
I think simplifying the environment lights, shadows, etc. would be the best way here

closed 2024-09-28 by reza-shahriari