Cosys-Lab/Cosys-AirSim #27: Docker for Cosys Airsim

Gavin2417, 2024-09-09:
Hi, is there a docker for this cosys airsim, I saw there is one for airsim but it used ubuntu 18.04?

Gavin2417, 2024-09-10:
> Hi, the docker folder is left-over from the Microsoft AirSim repo and has not been updated indeed. It is on our to-do list.

I see, and when will the docker updated and release with newer version of Ubuntu for the cosys airsim.

Gavin2417, 2024-09-10:
Is there an ETA for the release?

WouterJansen, 2024-09-10:
Cosys-AirSim is a research project for our group and as such we currently do not have an ETA on a future release. This is all dependent on available time and resources internally. Of course, this being an open-source project, we very much welcome the community to add updates and new features such as this 😃

WouterJansen, 2024-10-03:
We have updated the docker [tutorial](https://cosys-lab.github.io/docker_ubuntu/) and associated [scripts](https://github.com/Cosys-Lab/Cosys-AirSim/tree/main/docker) to work with current versions of Docker and use new official base images by Unreal Engine to use with runtime binaries or use the simulator from source.