Cosys-Lab/Cosys-AirSim #28: UE5.3 Issue

vincentlincoln, 2024-09-13:
[image]

Hi, I am new with cosys airsim, and i have an issue with UE5.3.2.

I have theses warnings too: 
[image]

vincentlincoln, 2024-09-13:
Does it supports 5.3.2?

WouterJansen, 2024-09-16:
Are you using from source or the prebuilt plugins? The prebuilt plugin from the releases page only supports 5.4. 

If you are using from source you should use the [5.3 branch](https://github.com/Cosys-Lab/Cosys-AirSim/tree/5.3.2) as that one should work better for Unreal 5.3

WouterJansen, 2024-09-16:
Strange, I just did a fresh checkout of the 5.3 branch with Unreal 5.3.2 and got a working Blocks environment without any build errors.

Visual Studio 2022 14.41.34120
Windows 10.0.22621.0 SDK
DotNet SDK version: 6.0.302

WouterJansen, 2024-10-04:
We made a small update and new release, I pushed the same changes and fixes to the 5.2 and 5.3 branches.

I did another clean install and did not run into any problems again.
- Unreal 5.3.2
- CMake 3.24.2
- Visual Studio 22 (17.11.4)
- MSVC 19.41.34120.0
- Windows SDK version 10.0.22621.0

 
I did however have an issue with building the RPC library while running the _build.cmd_ script. It gave me ` fatal error C1090: PDB API call failed, error code '3'`. The reason was related to my Dropbox client running in the background and causing a compile error. So I would suggest to always close programs like Dropbox, Google Drive, Onedrive, IDEs and other file checking systems while running the _build.cmd_ script. Could you try to run the _build.cmd_ script again and see if you run into any errors there? Make sure to run _clean.cmd_ first and see that the following folders are gone to trigger a proper rebuild:
- external
- AirLib/deps
- AirLib/lib/*
- Airlib/temp
- Unreal/Plugins/AirSim/AirLib
- Unreal/Environments/YOURENVIRONMENT/Plugins
- Unreal/Environments/YOURENVIRONMENT/DerivedDataCache
- Unreal/Environments/YOURENVIRONMENT/Intermediate

and make sure _build.cmd_ completes without errors.

vincentlincoln, 2024-10-04:
Thank you for your help. Last week i tried with 5.2 and it works well. (I didn't retry with 5.3)

WouterJansen, 2024-10-04:
Given https://github.com/Cosys-Lab/Cosys-AirSim/issues/36 is similar I will close this issue and keep track of the other one for 5.3 build and compile problems.

closed 2024-10-04 by WouterJansen