Cosys-Lab/Cosys-AirSim #30: UE5.2.1 Crashing

batc8t, 2024-09-18:
i have an issue with UE5.2.1 , with the blocks environment , when i run the ros2 wrapper Unreal engine crashes with the following error

Assertion failed: !bPostTickComponentUpdate [File:D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\LevelTick.cpp] [Line: 887]

UnrealEditor_Engine
UnrealEditor_Engine
UnrealEditor_AirSim!UAirBlueprintLib::DrawPoint() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirBlueprintLib.cpp:116]
UnrealEditor_AirSim!UnrealLidarSensor::getPointCloud() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\UnrealSensors\UnrealLidarSensor.cpp:215]
UnrealEditor_AirSim!msr::airlib::LidarSimple::updateOutput() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\sensors\lidar\LidarSimple.hpp:100]
UnrealEditor_AirSim!msr::airlib::LidarSimple::update() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\sensors\lidar\LidarSimple.hpp:51]
UnrealEditor_AirSim!msr::airlib::UpdatableContainer<msr::airlib::SensorBase *>::update() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\UpdatableContainer.hpp:62]
UnrealEditor_AirSim!msr::airlib::SensorCollection::update() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\sensors\SensorCollection.hpp:86]
UnrealEditor_AirSim!msr::airlib::MultiRotorPhysicsBody::updateSensorsAndController() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\vehicles\multirotor\MultiRotorPhysicsBody.hpp:90]
UnrealEditor_AirSim!msr::airlib::FastPhysicsEngine::updatePhysics() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\physics\FastPhysicsEngine.hpp:114]
UnrealEditor_AirSim!msr::airlib::FastPhysicsEngine::update() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\physics\FastPhysicsEngine.hpp:50]
UnrealEditor_AirSim!msr::airlib::World::update() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\physics\World.hpp:53]
UnrealEditor_AirSim!msr::airlib::World::worldUpdatorAsync() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\physics\World.hpp:152]
UnrealEditor_AirSim!common_utils::ScheduledExecutor::executorLoop() [D:\AIRSIMSTUFF\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\common\common_utils\ScheduledExecutor.hpp:204]
UnrealEditor_AirSim!std::thread::_Invoke<std::tuple<void (__cdecl common_utils::ScheduledExecutor::*)(void),common_utils::ScheduledExecutor *>,0,1>() [C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.41.34120\INCLUDE\thread:61]
ucrtbase
kernel32
ntdll  

or the following error

Assertion failed: !bRenderStateDirty [File:D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\Components\ActorComponent.cpp] [Line: 1816] Failed to route CreateRenderState_Concurrent (LineBatchComponent /Engine/Transient.LineBatchComponent_6)

UnrealEditor_Engine
UnrealEditor_Engine
UnrealEditor_Engine
UnrealEditor_UnrealEd
UnrealEditor_UnrealEd
UnrealEditor
UnrealEditor
UnrealEditor
UnrealEditor
UnrealEditor
UnrealEditor
kernel32
ntdll

i have reinstalled Unreal engine, reinstalled VS, and Cosys-Airsim and rebuilt everything still getting the errors

WouterJansen, 2024-09-18:
Hi, can you share your airsim settings.json file?

WouterJansen, 2024-09-18:
Can you try turning off _DrawDebugPoints_ for the Lidar sensors?

batc8t, 2024-09-18:
turning off DrawDebugPoints  alone didnt work , however removing the  Resolution,  RotationsPerSecond, MeasurementsPerCycle, and turning off DrawDebugPoints seems to be working. i havent had a crash in 30 mins , before it was crashing as soon as i start the ros2 wrapper.  here is my new Lidar settings: 
"LidarA": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": false
}
i also removed the PassiveEchoBeacons and Beacons

petertheprocess, 2024-09-21:
any fix or potential clue on that? I have the same issue, especially when switching viewport.

batc8t, 2024-09-30:
in my case it was because it was trying to draw the debug lines even though it was set to false in the settings. it kept on breaking at this line in Visual studio, AirBluePrintLib.cpp -> DrawPoint function -> LineBatcher->DrawPoint(Position, Color.ReinterpretAsLinear(), Size, DepthPriority, PointLifeTime); after commenting it out and also these lines in UnrealLidarSensor.cpp -> UpdatePose() -> 	//UAirBlueprintLib::DrawPoint(actor_->GetWorld(), sensor_position, 5, FColor::Black, false, draw_time_);
	//FVector sensor_direction = Vector3rToFVector(VectorMath::rotateVector(VectorMath::front(), sensor_reference_frame_.orientation, 1));
	//UAirBlueprintLib::DrawCoordinateSystem(actor_->GetWorld(), sensor_position, sensor_direction.Rotation(), 25, false, draw_time_, 10);
it no longer crashes. although im not sure if this will affect the lidar data as i havent tested it extensively so far everything looks normal though.

WouterJansen, 2024-09-30:
Can you share your settings.json as well please? I will take a look.

batc8t, 2024-09-30:
[settings.json](https://github.com/user-attachments/files/17186075/settings.json)

closed 2024-09-30 by WouterJansen

WouterJansen, 2024-09-30:
I think to have fixed this, will be part of the next build. Let me know if still an issue.

commit ec143dfb (2024-09-30): Set drawing sensor to false by default to avoid startup issues to hopefully fix #30.
commit 6966f35d (2024-10-04): Set drawing sensor to false by default to avoid startup issues to hopefully fix #30.

(cherry picked from commit ec143dfbcf6a1179996fe3695a88456eea8e2f11)
commit 20d2d742 (2024-10-04): Set drawing sensor to false by default to avoid startup issues to hopefully fix #30.

(cherry picked from commit ec143dfbcf6a1179996fe3695a88456eea8e2f11)