Cosys-Lab/Cosys-AirSim #31: GpuLidar Multirotor mode

reza-shahriari, 2024-09-24:
Hi, 
why is it not possible to use GPU LIDAR in multirotor mode?
Is there any development needed to make it work?
would you tell me how much faster it is compared to common LIDAR? 
Thanks

WouterJansen, 2024-09-24:
Unfortunately due to the way AirSim is structured, the multirotor code is run inside the airlib library and is not running as far as I know on the main simulation thread with access to the render thread. So the way we developed the GPU LiDAR it crashes when trying to access the lidar data on the render thread when using a multirotor vehicle. We currently do not have plans to fix it as it would require a refactor of our code. 

We also heavily optimized the normal LiDAR compared to Microsoft AirSim and made it more realistic. The biggest difference is that it does not use the multi-layer annotation system and cannot calculate intensity data. But performance-wise you should be fine with a decent CPU.

reza-shahriari, 2024-09-25:
Thank you very much for your quick and detailed response!

closed 2024-09-25 by reza-shahriari