Cosys-Lab/Cosys-AirSim #33: gpu lidar crash

reza-shahriari, 2024-09-28:
Hi,
this is my settings.json file:
```
{
  "SettingsVersion": 2.0,
  "SimMode": "ComputerVision",
  "Vehicles": {
      "CVSensor": {
          "VehicleType": "ComputerVision",
          "AutoCreate": true,
          "Sensors": {
            "gpulidar1": {
              "SensorType": 8,
              "Enabled" : true
          }
          }
      }
  }
}

```
when I run the game I see this error:
[image]
changing settings and adding more details to GPU lidar (as mentioned [here](https://cosys-lab.github.io/gpulidar/)) not change the problem! 
even using Car simulation leads to the same problem. when I Disable(Enabled: false) the GPU lidar sensor everything works perfectly

WouterJansen, 2024-09-30:
Can you share you settings.json file please? It seems to be tring to load a material CSV file for the intensity generation of GPU LiDAR and something is going wrong there.

reza-shahriari, 2024-09-30:
> Can you share you settings.json file please?

this is my setting.json file:
{
  "SettingsVersion": 2.0,
  "SimMode": "ComputerVision",
  "Vehicles": {
      "CVSensor": {
          "VehicleType": "ComputerVision",
          "AutoCreate": true,
          "Sensors": {
            "gpulidar1": {
              "SensorType": 8,
              "Enabled" : true
          }
          }
      }
  }
}

for material.csv I have an empty csv file in the Documents/airsim folder

reza-shahriari, 2024-10-01:
Thanks.
deleting the material.csv solved my problem

reza-shahriari, 2024-10-01:
[image]
The crash problem solved and its not crashing now but lidar GPU results is almost always just a point!
```
    def parse_gpulidarData(self):
        points = np.array(self.data.point_cloud, dtype=np.dtype('f4'))
        points = np.reshape(points, (int(points.shape[0] / 5), 5))
        xyz = points[:, 0:3]
        self.points = xyz

def visualize(self):
        print("visualizing data ...")
        
       # Convert NED (AirSim) to ENU coordinate system
        x = self.points[:, 0]
        y = -self.points[:, 1]  # Flip Y axis
        z = -self.points[:, 2]  # Flip Z axis

        # Verify the number of points
        print(f"Total points: {x.shape[0]}")  

        # RGB extraction (assuming the rgb array is already defined)
        # Normalize RGB values to the [0, 1] range for matplotlib
        rgb_normalized = self.rgb / 255.0

        # Plotting the Lidar data with colors
        fig = plt.figure()
        ax = fig.add_subplot(111, projection='3d')

        # Scatter plot with colors
        scatter = ax.scatter(x, y, z, c=rgb_normalized, marker='o')

        # Set labels
        ax.set_xlabel('X')
        ax.set_ylabel('Y')
        ax.set_zlabel('Z')

        # # Adjust the aspect ratio for better visualization
        # max_range = np.array([x.max() - x.min(), y.max() - y.min(), z.max() - z.min()]).max() / 2.0
        # mid_x = (x.max() + x.min()) * 0.5
        # mid_y = (y.max() + y.min()) * 0.5
        # mid_z = (z.max() + z.min()) * 0.5

        # ax.set_xlim(mid_x - max_range, mid_x + max_range)
        # ax.set_ylim(mid_y - max_range, mid_y + max_range)
        # ax.set_zlim(mid_z - max_range, mid_z + max_range)

        # Display the plot
        plt.show()
```
visualizing data ...
Total points: 32768

reza-shahriari, 2024-10-01:
settings file:
```
{
  "SettingsVersion": 2.0,
  "SimMode": "ComputerVision",
  "Vehicles": {
      "Drone1": {
          "VehicleType": "ComputerVision",
          "AutoCreate": true,
          "Sensors": {
            "gpulidar1": {
            "SensorType": 8,
            "Enabled" : true,
            "Range":150,
            "NumberOfChannels": 16,
            "RotationsPerSecond": 10,
            "PointsPerSecond": 100000,
            "X": 0, "Y": 0, "Z": -1,
            "Roll": 0, "Pitch": 0, "Yaw" : 0,
            "VerticalFOVUpper": -15,
            "VerticalFOVLower": -25,
            "HorizontalFOVStart": -20,
            "HorizontalFOVEnd": 20,
            "DrawDebugPoints": true
          }
          }
      }
  }
}
```

WouterJansen, 2024-10-04:
Hi, I get the same results. One thing, _RotationsPerSecond_ is not an argument for the LiDARs in Cosys-AirSim. We use _MeasurementsPerCycle_ to define the horizontal sample density. See the documentation on the [LiDAR](https://cosys-lab.github.io/lidar/) and [GPULiDAR](https://cosys-lab.github.io/gpulidar/) for all the arguments that can be used.

It seems you are running into an edge case that is causing a bug when to reset the pointcloud data. This is due to you using a negative _HorizontalFOVStart_ argument. I will fix this in a future release. For now you can either switch to the normal LiDAR sensor which does not have this bug or do a little change to your settings to prevent having to use a negative  _HorizontalFOVStart_. For example in your case you want to have a 40 degrees FOV from -20 degrees to 20 degrees, you have to set it to _Yaw_ : -20, _HorizontalFOVStart_: 0 and_HorizontalFOVEnd_: 40.

reza-shahriari, 2024-10-05:
Thanks for the answer.
changing setting to this worked nice for me:
```
{
  "SettingsVersion": 2.0,
  "SimMode": "ComputerVision",
  "Vehicles": {
    "Drone1": {
      "VehicleType": "ComputerVision",
      "AutoCreate": true,
      "Sensors": {
        "gpulidar1": {
          "SensorType": 8,
          "Enabled": true,
          "External": false,
          "NumberOfChannels": 50,
          "Range": 100,
          "MeasurementsPerCycle": 2048,
          "RotationsPerSecond": 10,
          "Resolution": 2048,
          "X": 0,
          "Y": 0,
          "Z": -0.5,
          "Roll": 0,
          "Pitch": 0,
          "Yaw": 0,
          "VerticalFOVUpper": 30,
          "VerticalFOVLower": -30,
          "HorizontalFOVStart": 0,
          "HorizontalFOVEnd": 360,
          "DrawDebugPoints": false,
          "DrawMode": 0,
          "IgnoreMarked": true,
          "GroundTruth": true,
          "InstanceSegmentation": true,
          "Annotation": "",
          "GenerateIntensity": false,
          "rangeMaxLambertianPercentage": 80,
          "rainMaxIntensity": 70,
          "rainConstantA": 0.01,
          "rainConstantB": 0.6,
          "DrawSensor": true
        }
      }
    }
  }
}
```

commit 52e39892 (2024-09-30): If material CSV file is empty or contains only empty line it will skip the creation. Fixed #33
commit d58b9ebb (2024-10-04): If material CSV file is empty or contains only empty line it will skip the creation. Fixed #33

(cherry picked from commit 52e398922a480f2a40deb05003f3672f59e7c0d3)
commit 8769c933 (2024-10-04): If material CSV file is empty or contains only empty line it will skip the creation. Fixed #33

(cherry picked from commit 52e398922a480f2a40deb05003f3672f59e7c0d3)
commit 87c2be02 (2024-11-25): Fixed GPULiDAR horizontal FOV scanning and resetting for none full 360 LiDARs. Fixes #33
closed 2024-10-07 by reza-shahriari