Cosys-Lab/Cosys-AirSim #34: Lidar unexpected results

reza-shahriari, 2024-09-28:
Hi,
sorry I could not figure out what my problem is with using lidar in a multirotor pawn. would you help me?
this is my environment:
[image]
and I am reading/visualizing the Lidar data like this:
```
data = client.getLidarData(self.drone_lidar_name,self.drone_name,)
points = np.array(data.point_cloud, dtype=np.dtype('f4'))
points = np.reshape(points, (int(points.shape[0]/3), 3))
 # Extract X, Y, Z coordinates
  x = points[:, 0]
  y = points[:, 1]
  z = points[:, 2]
  # Plot the point cloud
  fig = plt.figure()
  ax = fig.add_subplot(111, projection='3d')
  ax.scatter(x, y, z)
  # Set labels
  ax.set_xlabel('X Label')
  ax.set_ylabel('Y Label')
  ax.set_zlabel('Z Label')
  plt.show()
```
and this is the result:
[image]

this is my lidar settings if it helps:
```
 "Lidar1":{
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "RotationsPerSecond": 1,
          "MeasurementsPerCycle": 1000,
          "X": 0, "Y": 0, "Z": -1,
          "Roll": 0, "Pitch": 0, "Yaw" : 0,
          "VerticalFOVUpper": 0,
          "VerticalFOVLower": 360,
          "HorizontalFOVStart": 0,
          "HorizontalFOVEnd": 360,
          "Range":150,
          "DataFrame":"SensorLocalFrame",
          "DrawDebugPoints": true
        }
```
I don't think reading the data like this is a good idea! I just saw the `drone_lidar.py` and implemented this.

WouterJansen, 2024-09-30:
Hi, first of all I don't think setting vertical FOV to be between 0 and 360 degrees is a good idea or will work as intended. You should keep these values between -90 and 90 degrees. the _Dataframe_ argument is also no longer in use in Cosys-AirSim and will always be set to the sensor frame by default. 

The sensor data output is in NED format as most things are in AirSim. You need to flip the sign of the Y and Z coordinates to have a correct right-handed coordinate scheme plot.

reza-shahriari, 2024-09-30:
thanks for the answer.
I have changed the settings.json file as you suggested and changed the visualization. 
its better now but not as well as I wanted... may be I am wrong...

closed 2024-09-30 by reza-shahriari