Cosys-Lab/Cosys-AirSim #36: unreal 5.3.2 airsim problem

ZivniR, 2024-10-01:
hello,

when i built the game with 5.3.2 cosys-AirSim version the blocks worked perfectly open game and run

but when i try to use on my own project i get errors in Eigen every time like if i take an old version which suitable for 5.2 and add this plug in to the project. (the unreal-editor didnt open)

i try to use 5.4 it open the editor but cant run because of the other issue there was open here.

could you give me an suggestion what i am doing wrong?

thank you very much!

WouterJansen, 2024-10-01:
You are building the plugin from source right? As we dont have a prebuilt version for 5.3.

Can you post the exact error you get? 

Can you check if your uproject file is similar to the [one used in Blocks](https://github.com/Cosys-Lab/Cosys-AirSim/blob/5.3.2/Unreal/Environments/Blocks/Blocks.uproject)? 

Follow the guide [here](https://cosys-lab.github.io/unreal_custenv/#step-by-step-instructions-when-using-cosys-airsim-from-source-build) for the step-by-step guide for custom environments.

ZivniR, 2024-10-01:
yes i check and make it as same as possible here is the returning error i deal with:
[image]

WouterJansen, 2024-10-04:
We made a small update and new release, I pushed the same changes and fixes to the 5.2 and 5.3 branches.

I did a clean install and did not run into any problems.
- Unreal 5.3.2
- CMake 3.24.2
- Visual Studio 22 (17.11.4)
- MSVC 19.41.34120.0
- Windows SDK version 10.0.22621.0
 
I did however have an issue with building the RPC library while running the _build.cmd_ script. It gave me ` fatal error C1090: PDB API call failed, error code '3'`. The reason was related to my Dropbox client running in the background and causing a compile error. So I would suggest to always close programs like Dropbox, Google Drive, Onedrive, IDEs and other file checking systems while running the _build.cmd_ script. Could you try to run the _build.cmd_ script again and see if you run into any errors there? Make sure to run _clean.cmd_ first and see that the following folders are gone to trigger a proper rebuild:
- external
- AirLib/deps
- AirLib/lib/*
- Airlib/temp
- Unreal/Plugins/AirSim/AirLib
- Unreal/Environments/YOURENVIRONMENT/Plugins
- Unreal/Environments/YOURENVIRONMENT/DerivedDataCache
- Unreal/Environments/YOURENVIRONMENT/Intermediate

and make sure _build.cmd_ completes without errors.