Cosys-Lab/Cosys-AirSim #39: Object Detection Issue

vincentlincoln, 2024-10-07:
Hi,

Thank for your help the last time.

I am now trying the Object detection functionnality to annotate mesh with bounding box but nothing appear and even unwanted objects are annotated.

I am using the code from detection.py to detect static mesh actor with ID_names

For example :  the ID_name "StaticMeshActor_UAID_AC9_123" 

I tested multiples regex such as  "StaticMeshActor[\w]*", "StaticMeshActor.*" or even  "StaticMeshActor_.*_.*_.*$" 

but i always get the same issues

WouterJansen, 2024-10-08:
Can you be more specific please? Can you share more code that you are running as well as details about the other objects that it is listing?

vincentlincoln, 2024-10-08:
```python 
import cosysairsim as airsim
import cv2
import numpy as np 
import pprint

# connect to the AirSim simulator
client = airsim.VehicleClient()
client.confirmConnection()

# set camera name and image type to request images and detections
camera_name = "0"
image_type = airsim.ImageType.Scene

# set detection radius in [cm]
client.simSetDetectionFilterRadius(camera_name, image_type, 200 * 100) 
# add desired object name to detect in wild card/regex format
client.simClearDetectionMeshNames(camera_name, image_type)
client.simAddDetectionFilterMeshName(camera_name, image_type, "StaticMeshActor_UAID_*")

while True:
    rawImage = client.simGetImage(camera_name, image_type)
    if not rawImage:
        continue
    png = cv2.imdecode(airsim.string_to_uint8_array(rawImage), cv2.IMREAD_UNCHANGED)
    client.simClearDetectionMeshNames(camera_name, image_type)
    cylinders = client.simGetDetections(camera_name, image_type, annotation_name="StaticMeshActor_UAID_*")
    if cylinders:
        for cylinder in cylinders:
            s = pprint.pformat(cylinder)
            print("Cylinder: %s" % s)

            cv2.rectangle(png,(int(cylinder.box2D.min.x_val),int(cylinder.box2D.min.y_val)),(int(cylinder.box2D.max.x_val),int(cylinder.box2D.max.y_val)),(0,0,255),5)
            cv2.putText(png, cylinder.name, (int(cylinder.box2D.min.x_val),int(cylinder.box2D.min.y_val - 10)), cv2.FONT_HERSHEY_SIMPLEX, 2, (0,0,255),3)

    image_resized = cv2.resize(png, (0, 0), fx=0.25, fy=0.25)
    cv2.imshow('Image Redimensionnée', image_resized)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    # elif cv2.waitKey(1) & 0xFF == ord('c'):
    #     client.simClearDetectionMeshNames(camera_name, image_type)
    # elif cv2.waitKey(1) & 0xFF == ord('a'):
    #     client.simAddDetectionFilterMeshName(camera_name, image_type,  "StaticMeshActor_.*_.*_.*$")
cv2.destroyAllWindows()
``` 
Id_names of the unwanted detected meshes are for examples:

- StaticMeshActor_5
- RandomObject_C_1 (spawned from a blueprint class)
- Beacon_C_UAID_ABC123_123 (another blueprint)
- Fence01_01

I don't know it have an effect but I set "Using External Actor" option in World Settings.

WouterJansen, 2024-10-08:
I see the issue. The _annotation_name_ argument of the _simGetDetections()_ and other [object detection](https://cosys-lab.github.io/object_detection/) functions is for when using the [annotation camera](https://cosys-lab.github.io/annotation/).

This has more to do with which camera you want to detect the objects on and not the object filter.
To define the object filter you need to use _simAddDetectionFilterMeshName()_ function before running _simGetDetections()_ where you can use regex as well to filter out objects. So Similar to the cylinders in the original detection.py example, you should do:

```python
import cosysairsim as airsim
import cv2
import numpy as np 
import pprint

# connect to the AirSim simulator
client = airsim.VehicleClient()
client.confirmConnection()

# set camera name and image type to request images and detections
camera_name = "0"
image_type = airsim.ImageType.Scene

# set detection radius in [cm]
client.simSetDetectionFilterRadius(camera_name, image_type, 200 * 100) 
# add desired object name to detect in wild card/regex format
client.simAddDetectionFilterMeshName(camera_name, image_type, "StaticMeshActor_UAID_*")

while True:
    rawImage = client.simGetImage(camera_name, image_type)
    if not rawImage:
        continue
    png = cv2.imdecode(airsim.string_to_uint8_array(rawImage), cv2.IMREAD_UNCHANGED)
    client.simClearDetectionMeshNames(camera_name, image_type)
    detectedObjects = client.simGetDetections(camera_name, image_type)
    if detectedObjects:
        for detectedObject in detectedObjects:
            s = pprint.pformat(detectedObject)
            print("Cylinder: %s" % s)

            cv2.rectangle(png, (int(detectedObject.box2D.min.x_val), int(detectedObject.box2D.min.y_val)), (int(detectedObject.box2D.max.x_val), int(detectedObject.box2D.max.y_val)), (0, 0, 255), 5)
            cv2.putText(png, detectedObject.name, (int(detectedObject.box2D.min.x_val), int(detectedObject.box2D.min.y_val - 10)), cv2.FONT_HERSHEY_SIMPLEX, 2, (0, 0, 255), 3)

    image_resized = cv2.resize(png, (0, 0), fx=0.25, fy=0.25)
    cv2.imshow('Image Redimensionnée', image_resized)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    # elif cv2.waitKey(1) & 0xFF == ord('c'):
    #     client.simClearDetectionMeshNames(camera_name, image_type)
    # elif cv2.waitKey(1) & 0xFF == ord('a'):
    #     client.simAddDetectionFilterMeshName(camera_name, image_type,  "StaticMeshActor_.*_.*_.*$")
cv2.destroyAllWindows()
```

That said, I have no experience with the World Partition system of Unreal and the External Actor system so I do not know the effect this has on the naming of actors. Cause do note that the names you give to objects in your world in the editor do not necessarily mean that those are the names your objects will get. You can use the _client.simListSceneObjects()_ or _client.simListInstanceSegmentationObjects()_ to retrieve a list of names of all objects in the environment to check.

vincentlincoln, 2024-10-08:
Thank you for your answer.
But with your code, which I've just tested, I still get the same results. 

Both functions client.simListSceneObjects() or client.simListInstanceSegmentationObjects()  match the ID names displayed on Unreal Engine Outliner.

WouterJansen, 2024-10-08:
Do you get all the objects you want to find or are there also objects missing that should be detected?

vincentlincoln, 2024-10-08:
Yes , I got the objects i wanted to find , all the StaticMeshActor_UAID__ are listed. 

I also just did a new test and desactivated external actors and put some new object, which have the ID_name StaticMeshActor_1 and the regex StaticMeshActor_*. 

I still get the same issue.

WouterJansen, 2024-10-09:
First of all, the example code I posted above contains a error, it still contains `client.simClearDetectionMeshNames(camera_name, image_type)` in the loop causing all objects to be detected, this line should be placed above the `client.simAddDetectionFilterMeshName(camera_name, image_type, "StaticMeshActor_UAID_*")` once. 

After some digging around, I think I found the reason of the original issue though. In Unreal you have object names, which afaik cannot be changed from the editor and not as easy from C++. When using the editor, objects in the world outliner also have a label, the name you give them in the editor. This does not exist in packaged builds. However, the object detection code mostly uses the object names, not the label. So even if you set the object label, the underlying name might be different, causing issues with the object detection. You can see the object names in the world outliner:
[image]

Example, I have a set of objects starting with _cylinder_ in label and name, and a group of objects starting with _cylinder_ in name but changed the labels to start with _StaticMeshActor_UAID_. As you see here in the world outliner, all the objects have _Cylinder_ in the ID Name field.
[image]

The object detection will detect all of them when searching for _Cylinder*_ and nothing for _StaticMeshActor_UAID*_ cause the underlying name still contain_cylinder_.

`client.simAddDetectionFilterMeshName(camera_name, image_type, "Cylinder*") results in: 
`![image](https://github.com/user-attachments/assets/7feea97b-ca65-4a50-bbcd-9284b5b9da57)

So in your case I believe some of the objects that are still being detected, have a name that still contains _StaticMeshActor_UAID_ even though the label might be set differently. 

In the two main functions of the object detection system ([DetectionComponent.cpp](https://github.com/Cosys-Lab/Cosys-AirSim/blob/main/Unreal/Plugins/AirSim/Source/DetectionComponent.cpp) &[ObjectFilter.cpp](https://github.com/Cosys-Lab/Cosys-AirSim/blob/main/Unreal/Plugins/AirSim/Source/ObjectFilter.cpp)) we use mostly `actor->GetName()`. This could be changed to `actor->GetActorLabel()` or `actor->GetActorNameOrLabel()` but this likely will have other bad outcomes as well. You could try this out for yourself. 

So

vincentlincoln, 2024-10-09:
Hi,

Thanks for your answer it solved the problem. It is the client.simClearDetectionMeshNames(camera_name, image_type) in the loop causing the issue.

Concerning the names, I had understood correctly that object detection retrieves the Id Name (which is given by Unreal Engine) and not the label. 

The StaticMeshActor_UAID ID name is given by Unreal Engine when the External Actor option is activated.

[image]

closed 2024-10-09 by WouterJansen

ssaigarimella, 2025-07-24:
@WouterJansen @vincentlincoln by any chance do you know how to do 3D object detection and annotation with cosys-airsim?