Cosys-Lab/Cosys-AirSim #45: Hi, WouterJansen. How to fly my drone using keyboard in Cosys-Airsim/Block

pcompiler95, 2024-11-14:
Like the others, I ran the unreal engine in Cosys-airsim/Block environment. So I can see the drone, but it doesn't fly and not control using  
keyboard. So I am trying to fly the drone following the API.md in docs. But 'python install hello_drone.py' is not working.
I would love to hear from  the thankful AirSim supporters. Thank you.

ianare, 2024-11-14:
Here's what I did, on Linux and Python 3.12:

```
# install virtual environment
virtualenv venv

# activate the virtual environment
source ./venv/bin/activate

# install needed modules
pip install cosysairsim backports.ssl_match_hostname opencv-python

# download the test file
wget https://github.com/Cosys-Lab/Cosys-AirSim/raw/refs/heads/main/PythonClient/multirotor/hello_drone.py

# remove the useless import
sed -i 's/import setup_path//' hello_drone.py

# in another terminal, run airsim Blocks.sh

# run python file
python hello_drone.py
```

WouterJansen, 2024-11-18:
AFAIK you cannot control the drones in (Cosys-)AirSim with the keyboard. You need some form of remote control: https://cosys-lab.github.io/remote_control/#remote-control or use the API's to control it. 

Could you eleborate further on what is not working with the hello drone script? As others have tried it and it worked well. You can also not 'install' a python script as you show in your example code `python install hello_drone.py` above.

pcompiler95, 2024-11-19:
Thank sir. Could you help me one more?

pcompiler95, 2024-11-19:
Thank you for your consideration WouterJansen.
How to detect and recognize the images using AirSim drone. Of course. I ran the detection/detect.py. but AFAIK there is no image recognition method using python API. For example drone can detect images of character model in the airsim level. Can't recognize that the image is character, not other static mesh. In one word, how to recognize images using python API in airsim. 
Using Azure AI model or Hugging model or the other AI model?
Thank you. Good Awesome!

And the other question.
Now I can concurrently run the several python APIs in the Block project. Then, how can I make Unreal Engine.exe using the Block project and PythonClient.

Is there any other project can be executable? Not Block project else?

WouterJansen, 2024-11-19:
Other environments you have to create yourself, although there are many available on the Unreal marketplace/FAB and other places. You can use this tutorial on how to activate Cosys-AirSim in a different environment/Unreal project: https://cosys-lab.github.io/unreal_custenv/

nicolevowell5, 2024-11-19:
Thank you for your kindly guide. I am really appreciate about your help. Can I ask one more.
How to build the .exe with AirSim. There is no one guide about that. Please help me.

WouterJansen, 2024-11-19:
Building binaries is called packaging in Unreal terms, you can find more info here: https://dev.epicgames.com/documentation/en-us/unreal-engine/packaging-unreal-engine-projects

nicolevowell5, 2024-11-19:
Hi, @WouterJansen  Could we meet chat meeting?  skype: live:.cid.e22f452673fd0c01 
what do you prefer? telegram or what's app?

WouterJansen, 2024-11-19:
No sorry, I cannot help on that individual level.

closed 2024-11-19 by WouterJansen

nicolevowell5, 2024-11-19:
Oh, I see. Thank you for your consideration. @WouterJansen Really Thank  you.

nicolevowell5, 2024-11-19:
Hi @WouterJansen. Please teach me one more.
How to recognize images using python API in AirSim drone? 
In other word, Is there one way to Image recognition  in Airsim drone? not detection. using AI model with python API.
Thank you for your consideration. @WouterJansen.