Cosys-Lab/Cosys-AirSim #49: Add instructions for building packages to git

ianare, 2024-11-18:
Hi,

Would it be possible to add the build instructions for the release assets (Blocks, plugin) directly in git?

Even better if using GitHub actions, as this would allow anyone to reproduce a build in their forked repo.

I would like to help with this, but not sure where to start.

Thanks!

WouterJansen, 2024-11-19:
Hi, automated builds with actions or any other CI/CD pipelines are not likely at this point as I don't have the time or resources to set it up. Unless somebody else is willing to do so of course. 

The Blocks project I package manually on my workstation machines with the Editor itself through the GUI to create a packaged build. I choose Development as build format. 

For building the plugin, I use the build scripts of Unreal, for example on Windows I use: 
```
./RunUAT.bat BuildPlugin -Plugin=....\Cosys-AirSim\Unreal\Environments\Blocks\Plugins\AirSim\AirSim.uplugin -Package=...\airsimpluginpackagewin -Rocket -TargetPlatforms=Win64
```
On Linux I use:
```
sudo ./RunUAT.sh BuildPlugin -Plugin=.../Cosys-AirSim/Unreal/Environments/Blocks/Plugins/AirSim/AirSim.uplugin -Package=.../airsimpluginpackagelinux -Rocket -TargetPlatforms=Linux
```

I can add these somewhere in the documentation perhaps?

ianare, 2024-11-22:
Yes, it could be useful to have this documented somewhere.

I'll look into adding an automated build for the plugin.

Thanks

ianare, 2024-11-26:
I'm not seeing the `./RunUAT.sh` file in the repo, or am I missing something?

WouterJansen, 2024-11-28:
Sorry for the confusion. These are Unreal Engine build scripts found in your installation folder of Unreal Engine in the subfolder `\Engine\Build\BatchFiles`

ianare, 2024-11-28:
Looking into this a bit more, it's not easy to automate in Github, especially with the free tier.

The problem is due to licensing of UE and storage limitations on Github.

It would be great to have the instructions in the documentation though.

WouterJansen, 2024-11-29:
I will definitly add the instructions in a future update. I am concidering making a self-hosted github runner with UnrealEngine installed to make automated build on perhaps a weekly basis or so. But need to make some other choices first as well as for example which Unreal versions to support long-term etc.

ianare, 2024-11-29:
Yes automated builds would be great, and a time saver for you in the long run...

Regarding version support, I saw that recent changes to `main` breaks compilation on UE 5.4, not a problem per se but probably should be noted/documented somewhere.

WouterJansen, 2024-11-29:
Main is the development branch and is currently supporting 5.5, see the new 5.4 branch for this one.

commit 3c10a3aa (2024-11-29): Added instructions for packaging and building the plugin and/or environment/Unreal Project. Closes #49
commit f9eac504 (2024-11-29): Added instructions for packaging and building the plugin and/or environment/Unreal Project. Closes #49
commit 1716672b (2024-11-29): Added instructions for packaging and building the plugin and/or environment/Unreal Project. Closes #49
closed 2024-11-29 by WouterJansen