Cosys-Lab/Cosys-AirSim #52: miss label instance segmentation

reza-shahriari, 2024-11-21:
Hi,
look at the trees:
[image]
[image]

Why is this happening, and how can I prevent it?
it's not just in this frame I can see it in many other frames as well

WouterJansen, 2024-11-29:
Not sure, the annotation system uses proxy meshes and it seems there it is going wrong by choosing a different LOD or mesh to render the tree with. Could you share the asset?

reza-shahriari, 2024-12-02:
I am unsure if it is the best way but I have turned off all LOD changes and made them LOD0 all the time and I don't have the problem anymore!

ssaigarimella, 2024-12-12:
@reza-shahriari I'm having a similar issue, but my semantic segmentation frame is completely black. I am also using rural Australia environment, but I am unable to get the segmentation to look like what you have, can you please explain how you made the LOD changes? I am using Unreal Engine 5.2.1

ssaigarimella, 2024-12-12:
Update: Was able to fix this by setting the Default Level to be the chosen map, as shown here: [Changing the Default Level](https://dev.epicgames.com/documentation/en-us/unreal-engine/changing-the-default-level-of-an-unreal-engine-project)