Cosys-Lab/Cosys-AirSim #61: LiDAR issues with the Multirotor vehicle

bucknercsamuel, 2024-12-05:
I wanted to report/get help on multiple issues I am experiencing when using the LiDAR functionalities with a multirotor vehicle. 

(1) When I attempt to use GPULiDAR, I get an error saying this is incompatable with the multirotor model. Is this a bug, or a missing feature currently?

(2) I have a need to obtain a detailed pointcloud, and with this LiDAR design, this amounts to ~33k points for one full scan, which takes a non-negligible amount of time, and I also have my vehicle flying at a non-negligble speed. Consequently, the data from the scan does not appear to be position-adjusted for different sub-scans, so when flying the vehicle, I end up with a staggered pointcloud even when flying straight down and scanning a perpendicular plane. Is there any way to adjust for this without having to resort to hacky heuristics?

WouterJansen, 2024-12-05:
the GPULiDAR indeed does not work in its current implementation due to crashes related to the synchronization between game and render threads. An issue that has not been fixed or heavily investigated. Would likely require a rewrite to how the GPULiDAR works. 

As for the scanning, not entirely understanding the question. 33k points should be doable in real-time on most systems, but with the rotation speed of 1Hz it will have the realistic outcome of having subscans taken at different positions as the lidar rotates around during the motion. Is this what you mean?

bucknercsamuel, 2024-12-05:
Hey Wouter, thank you for the quick response. Regarding the second question, the scan rate for those 33k points, in terms of time it takes to to call `client.getLidarData`, is ~0.5 hz on my current computing hardware (aside: is this reasonable to expect from a modern gaming laptop?) -- this manifests as a large error in my resulting map construction due to the speed my vehicle is traveling at. I have also tested higher rotation/second values, however the scan rate still saturates at the aforementioned value, and the vertically-staggered pointcloud artifact is still observed.

bucknercsamuel, 2024-12-05:
Apologies, mistakenly closed the issue, it has been re-opened.

WouterJansen, 2024-12-20:
0.5 hz seems rather slow. 

I am still not sure I fully grap the problem. You have set the _RotationsPerSecond_ parameter to 1hz, so if you are traveling rather fast it will end up being indeed a staggered pointcloud as the simulation model realistically rotates the lidar around during that 1s period before the pointcloud gets updated.