Cosys-Lab/Cosys-AirSim #63: ListSceneObjects naming issue

Codaero, 2024-12-11:
I have some actors called TestCylinder of mesh type Cylinder as shown:
[image]

When I call listSceneObjects as follows, the names outputted are not the actor names:
`objs = client.simListSceneObjects("Cylinder[\w]*")`

This is the output (the TestCylinder1 is not named properly):
[image]

Suggestion:
Perhaps listSceneObjects should be able to return actor names? Additionally maybe an option to filter scene objects by actor names would be great to have.

WouterJansen, 2024-12-20:
It is a similar issue I believe as here: https://github.com/Cosys-Lab/Cosys-AirSim/issues/39#issuecomment-2401660593.

Codaero, 2025-01-26:
Thanks, not sure how else to get around it right now but will look into further.

closed 2025-01-26 by Codaero