Cosys-Lab/Cosys-AirSim #64: Question on Weather effects & textures

vincentlincoln, 2024-12-19:
Hi, 
I am using Airsim for drones. Currently there are no effect from weather, excepted some effect on camera. I am a beginner with Unreal Engine.
I know that there are Weather SFX uassets. How do you apply all of them on all objects & grounds?  

Do you have any guide or tutorial?

Thanks

ssaigarimella, 2024-12-19:
@vincentlincoln just checking, did you [follow this method to set the weather api? From the wiki, the weather APIs are disabled by default.](https://cosys-lab.github.io/apis/#weather-apis:~:text=retrieved%20using%20simGetWorldExtents().-,Weather%20APIs,all%20weather%20effects%20are%20disabled.%20To%20enable%20weather%20effect%2C%20first%20call%3A,-simEnableWeather(True))

vincentlincoln, 2024-12-20:
Yes , i did but the weather effects aren't apply on ground/object textures.I have snow, leafs effects.
I am using Cosys Airsim on my own map.

WouterJansen, 2024-12-20:
You need to add materials to your scene for it to read from the weather API i believe, I have not used these myself though.  
https://github.com/Cosys-Lab/Cosys-AirSim/tree/main/Unreal/Plugins/AirSim/Content/Weather/WeatherFX