Cosys-Lab/Cosys-AirSim #67: Error during build env Blocks

MaJiazhen8696, 2025-02-09:
UE 5.4.4
VS 2022 17.12.4
MSVC v143 latest 19.42.34436
Win10 
Airsim main on 2.9

I followed all the steps in the guide. When I run build.cmd there raised errors the same as  #22  and it worked. Then I run update_from_git  and open the blocks.sin and it shows up this

[image]

I chose OK and set Blocks as start up, change to development editor , Win64 and hit F5

> Build started at 22:23...
1>------ Build started: Project: Blocks, Configuration: Development_Editor x64 ------
1>Using bundled DotNet SDK version: 6.0.302
1>Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" BlocksEditor Win64 Development -Project="D:\Airsim\Airsim\Cosys-AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild -architecture=x64
1>Log file: C:\Users\MJZ\AppData\Local\UnrealBuildTool\Log.txt
1>Building BlocksEditor...
1>Using Visual Studio 2022 14.42.34436 toolchain (D:\Airsim\VS\IDE\VC\Tools\MSVC\14.42.34433) and Windows 10.0.22621.0 SDK (D:\Windows Kits\10).
1>EXEC : warning : Visual Studio 2022 compiler is not a preferred version
1>[Adaptive Build] Excluded from AirSim unity file: AirBlueprintLib.cpp, AirSim.cpp, AirSimCameraDirector.cpp, AirSimGameMode.cpp, DetectionComponent.cpp, LidarCamera.cpp, ManualPoseController.cpp, NedTransform.cpp, ObjectFilter.cpp, PawnEvents.cpp, PawnSimApi.cpp, PIPCamera.cpp, RenderRequest.cpp, TextureShuffleActor.cpp, UnrealImageCapture.cpp, WorldSimApi.cpp, RpcLibClientBase.cpp, RpcLibServerBase.cpp, FileSystem.cpp, ObstacleMap.cpp, SafetyEval.cpp, CarRpcLibClient.cpp, CarRpcLibServer.cpp, ComputerVisionRpcLibClient.cpp, ComputerVisionRpcLibServer.cpp, MultirotorApiBase.cpp, MultirotorRpcLibClient.cpp, MultirotorRpcLibServer.cpp, AnnotationComponent.cpp, ObjectAnnotator.cpp, DynamicBlockBeacon.cpp, DynamicRackBeacon.cpp, FiducialBeacon.cpp, PassiveEchoBeacon.cpp, TemplateBeacon.cpp, UWBBeacon.cpp, WifiBeacon.cpp, RecordingFile.cpp, RecordingThread.cpp, SimHUD.cpp, SimHUDWidget.cpp, DirectInputJoyStick.cpp, SimJoyStick.cpp, SimModeBase.cpp, SimModeWorldBase.cpp, UnrealDistanceSensor.cpp, UnrealEchoCommon.cpp, UnrealEchoSensor.cpp, UnrealGPULidarSensor.cpp, UnrealLidarSensor.cpp, UnrealMarLocUwbSensor.cpp, UnrealSensorFactory.cpp, UnrealSensorTemplate.cpp, UnrealWifiSensor.cpp, CarPawn.cpp, CarPawnApi.cpp, CarPawnSimApi.cpp, CarWheelFront.cpp, CarWheelRear.cpp, SimModeCar.cpp, ComputerVisionPawn.cpp, ComputerVisionPawnApi.cpp, ComputerVisionPawnSimApi.cpp, SimModeComputerVision.cpp, FlyingPawn.cpp, MultirotorPawnEvents.cpp, MultirotorPawnSimApi.cpp, SimModeWorldMultiRotor.cpp, SimModeSkidVehicle.cpp, SkidVehiclePawn.cpp, SkidVehiclePawnApi.cpp, SkidVehiclePawnSimApi.cpp, SkidWheel.cpp, WeatherLib.cpp
1>Determining max actions to execute in parallel (6 physical cores, 12 logical cores)
1>  Executing up to 6 processes, one per physical core
1>Using Parallel executor to run 83 action(s)
1>------ Building 83 action(s) started ------
1>[1/83] Compile [x64] SharedPCH.UnrealEd.Project.ValApi.Cpp20.cpp
1>D:\Airsim\UE\UE_5.4\Engine\Source\Runtime\Core\Public\Experimental\ConcurrentLinearAllocator.h(31): error C4668: '__has_feature' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>D:\Airsim\UE\UE_5.4\Engine\Source\Runtime\Core\Public\Experimental\ConcurrentLinearAllocator.h(31): error C4067: unexpected tokens following preprocessor directive - expected a newline
1>D:\Airsim\UE\UE_5.4\Engine\Source\Runtime\Engine\Classes\Engine\SkeletalMesh.h(930): warning C4996: 'FBoneMirrorInfo': FBoneMirrorInfo is deprecated. Please use UMirrorDataTable for mirroring support. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
1>Total time in Parallel executor: 19.30 seconds
1>Total execution time: 19.77 seconds
1>D:\Airsim\VS\IDE\MSBuild\Microsoft\VC\v170\Microsoft.MakeFile.Targets(44,5): error MSB3073: The command "D:\Airsim\UE\UE_5.4\Engine\Build\BatchFiles\Build.bat BlocksEditor Win64 Development -Project="D:\Airsim\Airsim\Cosys-AirSim\Unreal\Environments\Blocks\Blocks.uproject" -WaitMutex -FromMsBuild -architecture=x64" exited with code 6.
1>Done building project "Blocks.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
========== Build completed at 22:23 and took 20.776 seconds ==========

Thanks a lot ^_^

closed 2025-03-12 by MaJiazhen8696

liukeq, 2025-03-15:
I meet the same problem. Do you solve this?

MaJiazhen8696, 2025-03-15:
@liukeq  I downgrade my VS and it worked. Here's the link [https://stackoverflow.com/questions/72232850/how-to-install-specific-version-of-visual-studio-2022](url) Make sure you clean then build