Cosys-Lab/Cosys-AirSim #86: Cosys-Airsim cannot be packaged in UE5 editor

ClockZinc, 2025-05-15:
ENV INFO:
window10 UE5.5.4
cosys-airsim 3.3 windows plgins
I can play it in UE5 editor, but when I tried to packaged it, it turns to failed

[image]
error info
[image]

ClockZinc, 2025-05-15:
The above results were generated when running in the ​​CityParkEnvironmentCollec.uproject​​ with the following configuration:
""" json
{
	"FileVersion": 3,
	"EngineAssociation": "5.5",
	"Category": "",
	"Description": "",
	"Plugins": [
        {
            "Name": "AirSim",
            "Enabled": true
        }
    ]
}
"""
​​I disabled ChaosVehiclesPlugin because it caused the following errors/bugs:​
[image]

GateraGael, 2025-07-11:
Hello, were you able to fix this error ? If not which version of Unreal is recommended to package / build COsys-Airsim ?

WouterJansen, 2025-07-16:
https://github.com/Cosys-Lab/Cosys-AirSim/issues/102 seems to happen for others too