Cosys-Lab/Cosys-AirSim #88: Infrared camera always returns black frames (even after running create_ir_segmentation.py)

ssaigarimella, 2025-05-23:
**Operating System: Ubuntu 22.04 LTS
Unreal Engine version: 5.2.1
Cosys-AirSim version:[ 5.2.1 ]**(https://github.com/Cosys-Lab/Cosys-AirSim/releases/tag/5.2-v3.3)

Description
I’m using Cosys-AirSim on Ubuntu 22.04 + Unreal Engine 5.2.1. The IR camera (ImageType = 7) in Multirotor mode always returns a flat black image.

I've checked Instance segmentation works correctly; the mesh objects have a unique name and ID, and show up with unique colors:

[image]

IR frames remain all zeros in both the ROS2 viewer and a standalone OpenCV viewer.

I look at the [documentation on Infrared Cameras](https://cosys-lab.github.io/Cosys-AirSim/InfraredCamera/) and I noticed that the scripts `PythonClient/computer_vision/create_ir_segmentation_map.py` and `PythonClient/computer_vision/capture_ir_segmentation.py` do not exist in Cosys-AirSim; we need mainly `capture_ir_segmentation.py` to simulate the IR so I created a version that works with any UE5 world (pasted below). PythonClient/computer_vision/capture_ir_segmentation.py. When I run this and check the segmentation frame, it looks like the one below:

[image]

I've also confirmed my RGB scene image is working correctly:

[image]

However, I still get the black frames for the IR images.

I’ve experimented with pixels_as_float=True/False, with and without compression, and even swapped SimMode to ComputerVision – still black frames.

This is my `capture_ir_segmentation.py`:
`#!/usr/bin/env python3
import argparse
import sys
import random

import numpy as np
import setup_path 
import cosysairsim as airsim

def radiance(absoluteTemperature, emissivity, dx=0.01, response=None):
    """Compute spectral radiance (Planck × emissivity × camera response)."""
    wavelength = np.arange(8, 14, dx)
    c1 = 1.19104e8
    c2 = 1.43879e4

    # Planck’s law × emissivity × camera response (if provided)
    factor = response if response is not None else 1.0
    L = factor * emissivity * (
        c1 / ((wavelength**5) * (np.exp(c2 / (wavelength * absoluteTemperature)) - 1))
    )

    # integrate over λ to get scalar radiance
    if absoluteTemperature.ndim > 1:
        return L, np.trapz(L, dx=dx, axis=1)
    else:
        return L, np.trapz(L, dx=dx)

def get_new_temp_emiss_from_radiance(tempEmissivity, response):
    """
    Given [[name, T, eps],...], compute a [name, count] array
    where count ∈ [0,255] is the integrated radiance mapped to 0–255.
    """
    # reshape for vectorized radiance
    temps = tempEmissivity[:, 1].astype(float).reshape(-1, 1)
    ems   = tempEmissivity[:, 2].astype(float).reshape(-1, 1)

    # compute radiance → scalar values
    _, values = radiance(temps, ems, response=response)
    counts = ((values / values.max()) * 255).astype(np.uint8)

    # return [[name, count], ...]
    return np.hstack((
        tempEmissivity[:, 0].reshape(-1, 1),
        counts.reshape(-1, 1),
    ))

def main():
    p = argparse.ArgumentParser()
    p.add_argument("--settings", required=True,
                   help="(unused) path to settings.json; kept for compatibility")
    args = p.parse_args()

    # connect to AirSim
    client = airsim.MultirotorClient()
    client.confirmConnection()

    # 1) load camera response
    try:
        response = np.load("camera_response.npy")
        print("Loaded camera_response.npy")
    except Exception:
        print("camera_response.npy not found, defaulting to flat response")
        response = None

    # 2) list all scene objects
    object_names = client.simListSceneObjects()

    # 3) build mesh‐type roots by first TWO tokens, skip numeric seconds
    roots = []
    for name in object_names:
        parts = name.split("_")
        if len(parts) >= 2 and not parts[1].isdigit():
            mesh_root = f"{parts[0]}_{parts[1]}"
            if mesh_root not in roots:
                roots.append(mesh_root)

    if not roots:
        print("No meshes found to map. Exiting.")
        sys.exit(1)

    # 4) prepare random T & ε for each mesh‐type
    segIdDict = { root: root.lower() for root in roots }
    tempEmissivity = np.array([
        [root.lower(), random.uniform(285, 310), random.uniform(0.85, 0.98)]
        for root in roots
    ], dtype=object)

    # 5) compute new digital counts
    tempEmissivityNew = get_new_temp_emiss_from_radiance(tempEmissivity, response)

    # 6) reset all IDs to zero
    if not client.simSetSegmentationObjectID(r"[\w]*", 0, True):
        # fallback: clear per‐mesh
        for mesh in object_names:
            client.simSetSegmentationObjectID(mesh, 0, False)

    # 7) assign each mesh‐instance individually
    reassigned = 0
    for root, lower in segIdDict.items():
        # lookup the count for this mesh‐class
        idx = np.where(tempEmissivityNew[:, 0] == lower)[0]
        if idx.size == 0:
            continue
        objectID = int(tempEmissivityNew[idx[0], 1])

        # for every object name that begins with root + "_"
        for mesh_name in object_names:
            if mesh_name.startswith(root + "_"):
                client.simSetSegmentationObjectID(mesh_name, objectID, False)
                reassigned += 1

    print(f"Reassigned {reassigned} object IDs")

if __name__ == "__main__":
    main()
`

I run it using `python3 create_ir_segmentation_map.py --settings ~/Documents/AirSim/settings.json`.

This is my` settings.json:`

`{
  "SeeDocsAt": "https://cosys-lab.github.io/settings/",
  "SettingsVersion": 2,
  "ClockSpeed": 1.0,
  "ApiServerPort": 41451,
  "ViewMode": "FlyWithMe",
  "SimMode": "Multirotor",
  "Vehicles": {
    "Drone1": {
      "VehicleType": "SimpleFlight",
      "X": 5,
      "Y": -5,
      "Z": 0,
      "Yaw": 0,
      "Cameras": {
        "front_center": {
          "X": 0.35,
          "Y": 0,
          "Z": 0.1,
          "Pitch": 10.0,
          "Roll": 0.0,
          "Yaw": 0.0,
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 1280,
              "Height": 720,
              "FOV_Degrees": 90,
              "ForceUpdate": true
            },
            {
              "ImageType": 1,
              "Width": 1280,
              "Height": 720,
              "FOV_Degrees": 90,
              "ForceUpdate": true
            },
            {
              "ImageType": 5,
              "Width": 1280,
              "Height": 720,
              "FOV_Degrees": 90,
              "ForceUpdate": true
            },
            {
              "ImageType": 7,
              "Width": 1280,
              "Height": 720,
              "FOV_Degrees": 90,
              "ForceUpdate": true
            }
          ]
        }
      }
    }
  }
}`

**Expected behavior**
Infrared frames should show non‐zero grayscale values (thermal contrast) once the segmentation IDs have been mapped, matching the behavior in “ComputerVision” mode and the official AirSim demos.

Actual behavior
Flat black image; all pixels are zero, min/max always 0.

**Request**
1. Is there a known issue with IR only running in Cosys-AirSim?
2. How can I further debug what the IR render pass is doing (or not doing) in Unreal?
3. Are there any plugin flags or code changes required to enable live IR in Multirotor/Car/ComputerVision mode?

Thank you for any pointers!

kennn3334-afk, 2025-11-21:
Any update on this issue, I am facing the same issue.

tuwi84, 2025-12-17:
Having same issue. Trying to solve this. It seems to be csv not properly loading as when rendering ir and seg seperately i can see for 1 frame ue5 compilation window if shader is not compiled. So the issue is pbl in somwhere reading .csv.