Cosys-Lab/Cosys-AirSim #107: Problem with segmentation API

MahmoudAmer96, 2025-07-25:
For the past three days, I’ve been attempting to control the segmentation colors in AirSim using Python, but unfortunately without success. The same code previously worked flawlessly with the original AirSim plugin on Unreal Engine 4.27.

For example, the following script is intended to assign a segmentation ID of 42 to all objects using a wildcard pattern. However, despite receiving no errors and getting confirmation from the client, no visible changes occur in the segmentation images. All objects remain unaffected. Even when specifying individual object names explicitly, the segmentation IDs are not updated.

import time  
import cosysairsim as airsim

# Connect to AirSim
client = airsim.MultirotorClient()
client.confirmConnection()

SEGMENTATION_ID = 42
print(f"[INFO] Setting all objects' segmentation ID to {SEGMENTATION_ID}")
client.simSetSegmentationObjectID(".*", SEGMENTATION_ID, True)

time.sleep(1)
print("[INFO] Done. All objects should now have the same color in the segmentation view.")

ssaigarimella, 2025-08-04:
Having the same issue. Any updates by any chance?

WouterJansen, 2026-11-10:
The regex issue was fixed by #174. Updates hitting the wrong mesh, or not being applied, are fixed for the next release. The example scripts are fixed too: they looked colors up by list position instead of object ID.

Use the names from `simListInstanceSegmentationObjects()`, not actor names. Look up a color with `colorMap[client.simGetSegmentationObjectID(name), :]`. Please retest with the next release.

closed 2026-11-10 by WouterJansen